chegewara/EspTinyUSB #125: USB host for CDC devices

elloza, 2022-11-15:
I would like to ask if it is possible with this library to receive data from one ESP32 that is sending it via USB (CDC data) on another ESP32-S2.

[ESP32]<----UART2USB--------USB---->[ESP32-S2]

I think that a USB host CDC on a would be necessary to be able to read the data coming from the ESP32.

Am I right? 
Could someone point me in the right direction?

Sorry for the ignorance.

Thank you very much.

Translated with www.DeepL.com/Translator (free version)

chegewara, 2022-11-15:
Well, its not that easy, thou its possible.

That UART to USB is not CDC-ACM, its is serial device using proprietary protocol.
As far as i know the driver is already implemented in esp-idf, at least for cp2102 it hink, but you have to know that cp2102, ch500, ft232 and other chips are using different protocols.
I think it would be easier (and maybe cheaper) to create UART to USB with esp32-s2, esp32-c3, esp32-s3.

I belive espressif already have the code which let to build such USB to UART, just cant find it now in espressif repos.

elloza, 2022-11-15:
Thank you so much for your quick answer @chegewara!

Yes. My idea is to read serial information coming from a USB port of a device that I can read from a COM port on Windows with the CP210x drivers, but I want to do this on another ESP32.

I'm sorry, I don't know if I am explaining this properly.

I was wondering if I could from an ESP32-S2 set up a USB host with which to read that information.

Thank you very much for your help!!!

chegewara, 2022-11-15:
Yes, i understand what you want to do, and like i said, cp210x is not using CDC protocol. It is using own protocol, based on CDC and it is possible to write such driver.
I cant promise, but maybe in the future i can add such support. In the meantime here are examples:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_vcp

elloza, 2022-11-15:
Ooook, Thank you so much for your help @chegewara!!

closed 2022-11-15 by elloza