chegewara/EspTinyUSB #128: Corrupted SD Filesystem on remount/reset

Chasbrot, 2023-01-01:
First Happy New Year!

Description:
When using the example sd_msc, everything works fine, i can create, update and delete files. BUT when I remount or reset the esp, the filesystem is corrupted. Windows cannot open or remove directories anymore, sometimes files are missing or parts of files are missing. When i access the SD card via the esp after unmount, the same strange behavior happens as in windows. The test script goes completly haywire.

I tried:
- Multiple different SD cards (mainly older models)
- Different FAT settings
- Windows write cache disabled in volume settings
- Controlled power supply, no ripple or voltage sags

After formating the SD cards are completly fine and work normally. Only after remount/reset this happens. 

Enviroment:
- Windows 10 22H2
- uC: https://www.waveshare.com/wiki/ESP32-S2-Pico
- similar sd breakout: https://www.amazon.de/SUNFOUNDER-Sockel-Arduino-R3-Mega-2560-Nano/dp/B07R6HQGS1
- cable length to sd ~ 5cm
- Used Library: lastest in the library manager
- Arduino IDE: legacy version (1.18.10)

chegewara, 2023-01-04:
Hi,
its possible there is missing fs sync command or something else.
I never tested it with windows and on linux i had similar issue few times, when i remove "pendrive" without unmounting it.

You can try to add this in you code, in main.ino:
```
void tud_msc_write10_complete_cb(uint8_t lun)
{
    if (disk_ioctl(0, CTRL_SYNC, NULL) != RES_OK)
    {
        Serial.printf("failed to sync\n");
    }
}
```

Chasbrot, 2023-01-06:
hi, thanks for the quick response. i also tried safe remove with windows, but that didn't help. After a while i just tired it with the target system (which is some kind of embedded "os") and it just worked. I am developing under windows, so my testing was with windows. Probably windows doing windows things. I tried your code (just copied it under the loop function):

```
C:\Users\Michael\Desktop\USBMonitor\USBMonitor.ino: In function 'void tud_msc_write10_complete_cb(uint8_t)':
USBMonitor:436:23: error: 'CTRL_SYNC' was not declared in this scope
     if (disk_ioctl(0, CTRL_SYNC, NULL) != RES_OK)
                       ^~~~~~~~~
C:\Users\Michael\Desktop\USBMonitor\USBMonitor.ino:436:23: note: suggested alternative: 'CARD_SDHC'
     if (disk_ioctl(0, CTRL_SYNC, NULL) != RES_OK)
                       ^~~~~~~~~
                       CARD_SDHC
USBMonitor:436:9: error: 'disk_ioctl' was not declared in this scope
     if (disk_ioctl(0, CTRL_SYNC, NULL) != RES_OK)
         ^~~~~~~~~~
C:\Users\Michael\Desktop\USBMonitor\USBMonitor.ino:436:9: note: suggested alternative: 'ioctl'
     if (disk_ioctl(0, CTRL_SYNC, NULL) != RES_OK)
         ^~~~~~~~~~
         ioctl
USBMonitor:436:43: error: 'RES_OK' was not declared in this scope
     if (disk_ioctl(0, CTRL_SYNC, NULL) != RES_OK)
                                           ^~~~~~
C:\Users\Michael\Desktop\USBMonitor\USBMonitor.ino:436:43: note: suggested alternative: 'ETS_OK'
     if (disk_ioctl(0, CTRL_SYNC, NULL) != RES_OK)
                                           ^~~~~~
                                           ETS_OK
```

chegewara, 2023-01-09:
I didnt post included headers:
```
#include "ff.h"
#include "diskio.h"
void tud_msc_write10_complete_cb(uint8_t lun)
{
    if (disk_ioctl(0, CTRL_SYNC, NULL) != RES_OK)
    {
        Serial.printf("failed to sync\n");
    }
}
```