chegewara/EspTinyUSB #129: ESP32-S3, reading file on SD Card through USB

RodEnry, 2023-01-10:
Hello @chegewara,
thanks for your great library.

I have an ESP32-S3 which is connected to a micro SD card module via SPI.
The library used for the SD is SdFat 2.

My goal

Cheers
Enry

chegewara, 2023-01-10:
Hi,
it may be between 500-800kB/s.

RodEnry, 2023-01-11:
@chegewara thanks!
In order to reach my goal, I'll need to use the Host MCC example?

Cheers
Enrico

RodEnry, 2023-01-12:
Hello, 
I don't know why but I can't access to the mass storage from the PC.
I can see this popup:

[image]

Any idea?
Cheers
Enry

chegewara, 2023-01-16:
Host example is to connect pendrive to esp32.

This error may be due to timeout, because of low transfer speed of USB from esp32.

fepalmytho, 2023-11-16:
Hi @RodEnry, i've been looking for days to find a way to do exactly the same than you. Make my ESP32-S3 appear as an USB stick, and reach by this way my SD CARD while connected to the PC (my SD card is a nand by the way, but connected and working as an SD card perfectly).
I would like to know if you can share with me your code, and if you managed to make it work properly.
I'm trying to make the sd_msc.ino code work, but i'm stuck with libraries missing, my knowledge is quite poor about how to set that and feed the correct information to visual studio code. For example, in the code, i had to comment #if CFG_TUD_MSC, as it makes me errors. I don't know what's the purpose of this line...
Thanks a million,
Frederic

fepalmytho, 2023-11-17:
Hi @RodEnry , 
Glad you're still interested in solving this problem, and so glad that i'm not "alone" to try to do something i guess "so basical"...
I'm quite surprised that something so "obvious" is not documented somewhere on a tutorial on Internet...
So, first of all, i don't have any code (for now) to share, as i'm just starting my researches.
Just to understand my goal, lets explain in a few lines my project.
I'm creating an "instrument" with a display. This instrument show informations, and i need to store them inside this project, as i don't want any wifi, bluetooth, GSM or whatever external connexion. I want this instrument to be totally autonomous.
So i decided to store data, a lot of data (as .txt files). I took the ESP32-S3 as it has a lot of SPIFFS memory, but not enough.
I've tried to implement a chip called Winbond W25QXX, but after researches, i was unable to make it work.
Problems with libraries, and must of all, i was limited by my skills. It seems to be that you can't read and write on this chips like an SD card, you need to write by « bites » or even more complicated, well, it was really not a plug and play device, and really not userfriendly.
So i decided to come back to an SD card (using an SD card Shield), which has a lot of capacity, and so easy to use with the dedicated code (library). But i didn’t like 2 things :
First, it’s so vibrating sensitive, that sometimes i had to plug and unplug the SD card to make it work. If not, i had times to times the message « card failed » or « not mounted ». Well, i could not use it like this. (I’ve tried to solder the SD card directly to a prototype PCB, but i twas uggly, not reliable as my soldering skills are not as robots, and even more, as far as i understood, it was impossible to make it communicate directly with the ESP32 via some (a lot of) GPIO pins, and i had absolutely need of the card shield).
Secondly, using a card shield, i didnt like the fact that a user could open the instrument, take away the SD card and plug it on a computer and see my data.
So i’ve found an SD NAND chip (the XTSD04GLGEAG), which is connected as the SD card Shield (MISO, MOSI, CS and CLK pins). This chip is really tiny, has a lot of capacity, and react as an SD card with the same libraries (tested and approved).
So now, here is where i’m starting to look for solutions to upload my files to this chip, in a easy way.
But be reassured, all my thoughts are exactly the sames using an SD card with a shield, as it seems to be your goal too.
I cannot unplug the chip and wire it to an USB cable to connect it directly to a computer. (Well at least i didnt find any information about this way).
So i decided to create a code that makes work in paralell an SD Card, and my Chip. So i could transfer my data on the Sd card from the computer, and then via the code, copy paste all the files to the chip. I know that it’s possible, but not user friendly, so i « paused » this way.
Second option, was to make the ESP32 appear as an USB mass Storage, and i found this famous TinyUSB code (the one from Chegewara that you’ve mentionned). It didnt work for 2 reasons:
First one, because it was looking for a library #include "sdusb.h", that i had to find in his files on github, and this library was looking for another, that i had to add, which was looking for another etc…it was abyssal and endless. And still showing code errors.
The second one, was that there was this condition #if CFG_TUD_MSC, which i absolutely don’t know what is for. I guess, like many other codes tested, that if  for helping setting regarding the MCU used ?!. But as soon as i don’t comment this line, all my code is grey. But even commented, i was struggled by the libraries.
Well, i decided to forget this code.
So today, i’m trying to understand this page :
https://www.reddit.com/r/arduino/comments/w4o8wn/esb32s3_nand_sd_storage_usb_msc_arduino_that/
And this one :
https://github.com/frankcohen/ReflectionsOS/blob/main/Devices/Hoober/Software/07USBFlashDrive/07USBFlashDrive.ino
I come back to you if i can move forward in any way.
If someone else reads thoses lines, we are more than happy to get help !
Frederic.

chegewara, 2023-11-17:
Sorry for no responding, but currently this library is on hold.
I am overworked and the library seems to be not popular. 

In the near future i will focus on some crypto AI app, which gives me $2000 a day with no risk and im not sure i get back to this library. Again, sorry for that.

fepalmytho, 2023-11-17:
Hi,
Thanks for updating us. So we are looking for other options.
Good luck with your no risk API.
Fred.