chegewara/EspTinyUSB #135: msc flashdisk compile error on Platformio with ESP32-S2

PaffAngryDragon, 2023-02-04:
When compiling on VSC+Platformio with ArduinoEspressif 4.1.0 I get errors for missing FS.h and SD.h, which could be fixed by simply adding #include <...> lines.  Next error is trickier: 
`
Compiling .pio\build\esp32-s2-saola-1\liba36\ESP32TinyUSB\host\msc\vfs\diskio_rawmsc.cpp.o
.pio/libdeps/esp32-s2-saola-1/ESP32TinyUSB/src/host/common/usb_host.cpp: In member function 'bool USBhost::init(bool)':
.pio/libdeps/esp32-s2-saola-1/ESP32TinyUSB/src/host/common/usb_host.cpp:72:5: error: 'const usb_host_client_config_t' has no non-static data member named 'is_synchronous'
     };
     ^
*** [.pio\build\esp32-s2-saola-1\liba36\ESP32TinyUSB\host\common\usb_host.cpp.o] Error 1`

It is because the struct declared is: 
`
    const usb_host_client_config_t client_config = {
        .is_synchronous = false,
        .max_num_event_msg = 5,
        .async = {
            .client_event_callback = _client_event_callback,
            .callback_arg = this
        }
    };
`  but the defined usb_host_client_config_t  declared in in esp32-s2 tools/sdk looks like this: 
`
/**
 * @brief USB Host Library Client configuration
 *
 * Configuration structure for a USB Host Library client. Provided in usb_host_client_register()
 */
typedef struct {
    usb_host_client_event_cb_t client_event_callback;   /**< Client's event callback function */
    void *callback_arg;                                 /**< Event callback function argument */
    int max_num_event_msg;                              /**< Maximum number of event messages that can be stored (e.g., 3) */
} usb_host_client_config_t;
Of course, I corrected it manually, but later run in some issues - perhaps because of  this.  I know that you are working on a new version - but until then, perhaps you could fix this issue?  Or even better, if you have some version of MSC perhaps I could try it?

I am trying to write a program ewhere the esp32-s2 would  have HIDUSB capability (done..) + become an USB MSC device with one FAT partition toward a PC and at the same time to have FATFS open on that partition (done, but separatelly of HIDUSB). When I try all three together I hit the wall for  2 dys now. All available USB librarys seems to have some issues with MSC +FAT on the internal flash of ESP32.
`

chegewara, 2023-02-04:
This library is working pretty good with MSC + FATFS, tested and confirmed by many users.

It is possible that API has been changed in arduino-esp32 (esp-idf) you are trying to build with.
What you can do is to edit host code, or even to delete it, because you dont need it.
You can also try to use my v2 library, but its still a bit messy, because ive been trying to write code compatible with arduino and esp-idf, which is not easy, because they are using different tinyusb ports (kind of sucks).

PaffAngryDragon, 2023-02-04:
Thank you for your superprompt answer! Tried deleting the host code, i.e. the arduino_tinyusb lib in \tools\sdk\esp32s2\include -  but it is a no-go , because I got error on dependency from your lib on: 

In file included from .pio/libdeps/esp32-s2-saola-1/ESP32TinyUSB/src/mscusb.h:2,
                 from include/flashdisk.h:2,
                 from src/main.cpp:10:
.pio/libdeps/esp32-s2-saola-1/ESP32TinyUSB/src/esptinyusb.h:7:10: fatal error: common/tusb_common.h: No such file or directory
 #include "common/tusb_common.h" - and this tusb_common.h I could not find in your library. And the before mentioned "usb_host_client_config_t" is also defined outside of your library. 
I am totally confused with this, I thought it would be an easy ride - but it looks like I am missing something. I included your lib with Platformio Library manager  - so there should be no missing parts.  Oh, that reminds me that flashdisk.h -was not in it , I had to get it separately from your Github. 
I am pretty much ruined mentally in these 2 days, so I have to ask your opinion: is it possible at all to have the internal fat partition of esp32-s2 mounted as a file system on esp32, AND be an USBMSC device to PC AT THE SAME TIME?  Because I have your super demo programs and they work, and the FFatdemo  from  arduinoespressif library FFAT examples works too, but only separately. When FFat starts the USB device disconnects from PC.