chegewara/EspTinyUSB #137: HID generic Computer Software

gary7530, 2023-02-16:
https://github.com/chegewara/EspTinyUSB/blob/master/examples/device/hid/generic/generic.ino

What software should I use on my computer?

chegewara, 2023-02-16:
You have to write own app or you may try to find one.
Its a bit like with BLE. This is not specified protocol, except that bytes are send both ways, but its up to you when to send data on both ends and what to do with it.

gary7530, 2023-02-23:
I am using this software on the computer side https://github.com/todbot/hidpytoy
I have successfully sent to ESP32S2 from my computer, but ESP32S2 failed to send to my computer.
computer->ESP32S2 
[image]
[image]

chegewara, 2023-02-25:
How HID descriptor looks like?

rayoffleon, 2023-05-14:
I have the same issue, i'm trying to replicate an MLA example from microchip (USB Custom), and can with HIDPyToy  send and
receive to a PIC 18F2550, but using the "generic" example from ESP32 S3 and the same HIDPyToy just can send and does not receive from ESP32, i think the problem or difference is that the PIC does not use the report ID, and the ESP32 uses it, is there a way to set a config descriptor that ignores the ID descriptors like in the MLA (report size = 0)? 

https://ww1.microchip.com/downloads/en/softwarelibrary/mla_v2018_11_26_windows_installer.exe

Here is the config descriptor from MLA that is working:

```
/* Configuration 1 Descriptor */
const uint8_t configDescriptor1[]={
    /* Configuration Descriptor */
    0x09,//sizeof(USB_CFG_DSC),    // Size of this descriptor in bytes
    USB_DESCRIPTOR_CONFIGURATION,                // CONFIGURATION descriptor type
    0x29,0x00,            // Total length of data for this cfg
    1,                      // Number of interfaces in this cfg
    1,                      // Index value of this configuration
    0,                      // Configuration string index
    _DEFAULT | _SELF,               // Attributes, see usb_device.h
    50,                     // Max power consumption (2X mA)
							
    /* Interface Descriptor */
    0x09,//sizeof(USB_INTF_DSC),   // Size of this descriptor in bytes
    USB_DESCRIPTOR_INTERFACE,               // INTERFACE descriptor type
    0,                      // Interface Number
    0,                      // Alternate Setting Number
    2,                      // Number of endpoints in this intf
    HID_INTF,               // Class code
    0,     // Subclass code
    0,     // Protocol code
    0,                      // Interface string index

    /* HID Class-Specific Descriptor */
    0x09,//sizeof(USB_HID_DSC)+3,    // Size of this descriptor in bytes
    DSC_HID,                // HID descriptor type
    0x11,0x01,                 // HID Spec Release Number in BCD format (1.11)
    0x00,                   // Country Code (0x00 for Not supported)
    HID_NUM_OF_DSC,         // Number of class descriptors, see usbcfg.h
    DSC_RPT,                // Report descriptor type
    HID_RPT01_SIZE,0x00,//sizeof(hid_rpt01),      // Size of the report descriptor
    
    /* Endpoint Descriptor */
    0x07,/*sizeof(USB_EP_DSC)*/
    USB_DESCRIPTOR_ENDPOINT,    //Endpoint Descriptor
    CUSTOM_DEVICE_HID_EP | _EP_IN,                   //EndpointAddress
    _INTERRUPT,                       //Attributes
    0x40,0x00,                  //size
    0x01,                        //Interval

    /* Endpoint Descriptor */
    0x07,/*sizeof(USB_EP_DSC)*/
    USB_DESCRIPTOR_ENDPOINT,    //Endpoint Descriptor
    CUSTOM_DEVICE_HID_EP | _EP_OUT,                   //EndpointAddress
    _INTERRUPT,                       //Attributes
    0x40,0x00,                  //size
    0x01                        //Interval
};
```
Sorry for my bad english, regards.

chegewara, 2023-05-15:
@rayoffleon In config descriptor you dont have reportID, it is in HID descriptor and it always present.
Again, reportId is assigned in constructor, so you can try code like this if you insist:
```
HIDgeneric dev(0);
```

chegewara, 2023-05-15:
And here is info from that app screen, which means you are sending to reportID=1. Thats why esp32 is receiving data

[image]

And app probably cant receive because of this (reportID=0):
[image]

chegewara, 2023-05-16:
I will try to investigate this, if i wont forgot it

rayoffleon, 2023-05-16:
Ok, i'm very interested in this, if i can help testing or something just tell me.

Thanks.