chegewara/EspTinyUSB #138: sd_mscerror

755516538, 2023-02-26:
Hello, I encountered an error when using the tinyusb example "sd_msc.ino". An error will occur if the file size of the SD card exceeds 512 bytes. Can you teach me how to repair it

chegewara, 2023-02-27:
Hi,
i dont understand what problem do you have, sorry.
If you describe it more, maybe i can help.

755516538, 2023-02-28:
Thank you for your reply.
Preparation for problem recurrence:
1.esp32s3
2. Text file larger than 512 bytes (000.txt)
3.sd_ Msc sample code
Step:
1. Modify the example to read and write the SD card correctly.
#define SD_ MISO  37
#define SD_ MOSI  39
#define SD_ SCK   38
#define SD_ CS    40
2. Burn the program to esp32s3
3. Copy "000. TXT" to esp32disk.
4. Unplug the esp32 and disconnect it from the computer
5. Insert esp32 into the computer and connect it to the computer, and open the USB flash drive
6. Open "000.TXT" in the USB flash drive, and some contents of the file are lost.
[image]

chegewara, 2023-02-28:
Its because you did not unmount device under system properly.
Im not sure, but this small snippet in your `.ino` may help to fix it:
```
#include "ff.h"
#include "diskio.h"
void tud_msc_write10_complete_cb(uint8_t lun)
{
    if (disk_ioctl(0, CTRL_SYNC, NULL) != RES_OK)
    {
        Serial.printf("failed to sync\n");
    }
}
```

755516538, 2023-02-28:
Ok, thank you very much for your help