chegewara/EspTinyUSB #140: My ESP32-S2 does not work when connected to unpowered USB 2.0 hub

UnaiUribarri-TomTom, 2023-05-16:
It works with USB 3.0 hubs and also with powered USB 2.0 hubs. After some debugging, it seems that my OS complains that there is not enough energy to feed ESP32-S2 because it is requesting 500mA and my unpowered hub can only provide 450mA.

I know that ESP32-S2 is happy with much less power than 500mA... how can I reduce the power requirement that my computer sees?

UnaiUribarri-TomTom, 2023-05-16:
Could be related with this line of code?

https://github.com/chegewara/EspTinyUSB/blob/f4d63153c0417922398de2fe43114b5224ff6def/src/usb_descriptors.cpp#L37

Would you accept a PR making that hard-coded 500 configurable?

chegewara, 2023-05-17:
Sure, i will accept reasonable PR.