chegewara/EspTinyUSB #142: Connect USB device and send HEX data

demenkovms, 2023-06-11:
Hello. I connect usb card-reader to ESP32-S2

D+ --> GPIO 20
D- --> GPIO 19
VCC --> v5
GND --> GND

how I can send HEX data for it and read byte answer? Thanks!

chiffacff, 2023-06-30:
No any idea?...

chegewara, 2023-07-03:
There is many ways to send HEX data
- USB CDC
- USB HID

all depends what you want to achieve and what app you have on the other side

demenkovms, 2023-07-03:
Thanks for answer. I tried to use CDC and HID from example,  but it's nothing result. I use this code:

`/**
 * Simple CDC device connect with putty to use it
 * author: chegewara
 * Serial - used only for logging
 * Serial1 - can be used to control GPS or any other device, may be replaced with Serial
 */
#include "cdcusb.h"
#if CFG_TUD_CDC
CDCusb USBSerial;

class MyUSBCallbacks : public CDCCallbacks {
    void onCodingChange(cdc_line_coding_t const* p_line_coding)
    {
        int bitrate = USBSerial.getBitrate();
        Serial.printf("new bitrate: %d\n", bitrate);
    }

    bool onConnect(bool dtr, bool rts)
    {
        Serial.printf("connection state changed, dtr: %d, rts: %d\n", dtr, rts);
        return true;  // allow to persist reset, when Arduino IDE is trying to enter bootloader mode
    }

    void onData()
    {
        int len = USBSerial.available();
        Serial.printf("\nnew data, len %d\n", len);
        uint8_t buf[len] = {};
        USBSerial.read(buf, len);
        Serial.write(buf, len);
    }

    void onWantedChar(char c)
    {
        Serial.printf("wanted char: %c\n", c);
    }
};

void setup()
{
    Serial.begin(115200);
    USBSerial.setCallbacks(new MyUSBCallbacks());
    USBSerial.setWantedChar('x');

    if (!USBSerial.begin())
        Serial.println("Failed to start CDC USB stack");

}

void loop() {

 if (Serial.available ())
  {
    String getCmd = Serial.readString();

    Serial.print("getCmd: ");
    Serial.println(getCmd);

    

    if (getCmd=="test") {

      Serial.println("Try send beep");

    uint8_t sendData [] = {0x4D, 0x00, 0x07}; //make beep of USB device

    USBSerial.write (sendData, sizeof(sendData));
    
Serial.println("send is ok");

    }

    Serial.println("*******************");
    

  }
}
` 

and on get "test" command usb-device are not make beep

chegewara, 2023-07-03:
Great
and how did you connect/wire esp32s2 board?

demenkovms, 2023-07-03:
like I wrote in first post:

USB D+ --> GPIO 20
USB  D- --> GPIO 19
USB  VCC --> v5
USB  GND --> GND

chegewara, 2023-07-03:
ok, but it is only half of it
you are using `Serial` in your code. do you have connected USB to UART converter too?

demenkovms, 2023-07-03:
sorry, are you about connecting esp32 to PC? - it’s over micro usb connector

demenkovms, 2023-07-03:
or I incorrect understand your question?…

chegewara, 2023-07-03:
With the code you posted you need to connect 2 USB cables to PC
- one for USB which you described
- one for UART which is requiring some USB to UART  converter like cp210x or ftdi, so you could use and to read `Serial`

You can also replace all `Serial` with `USBSerial` and to have one USB connection you have right now

demenkovms, 2023-07-04:
But, it's need connect usb device to ESP32...