chegewara/EspTinyUSB #143: Unmount a host msc usb

stetrx, 2023-06-19:
Hello.
I'm new to github and and it's the first time I'm approaching USB management.
I hope I'm asking for help correctly: I started from the example /examples/host/msc/msc.ino
I would like to be able to handle remounting a usb drive after unplugging it from the esp32s3 usb port (i'm using pin 19=d- and 20=d+). 
Starting from the example mentioned above , I added the management of the USB_HOST_CLIENT_EVENT_DEV_GONE event in 
 void _client_event_callback() inside usb_host.cpp module of library.
This is because otherwise the usb drive disconnect event was not detected inside void client_event_callback(const usb_host_client_event_msg_t *event_msg, void *arg) of msc.ino.

Once usb drive is disconnected, however, it is no longer remounted if I plug it again. 
I tried to implement the unmount function following what I found on https://github.com/chegewara/esp32-usb-host
but it does not work. I can only reset the esp32s3 to be able to mount usb drive again.

Could anyone help me and tell me how to fix the problem?
regards

chegewara, 2023-06-19:
Hi,
i dont remember what was the problem with usb host re-mount in esp-idf and if its fixed in v4.4.x.
Another thing you have to know is that you have to unmount file system when msc is unpluged and to destroy msc object.

Yes, this example is probably good starting point to do it
https://github.com/chegewara/esp32-usb-host/blob/main/examples/remote_pendrive/main/main.cpp

stetrx, 2023-06-20:
Hi.
Thank you very much for your quick reply.
I'm using https://github.com/espressif/arduino-esp32/releases/tag/2.0.9 as Arduino core (it's based on ESP-IDF 4.4.4).
What I found is that the library ESP32TinyUSB v.2.0.2 doesn't contain an unmount method.

Looking at other codes I tried to implement unmount method , but it doesn't work.
Currently, by using ESP32TinyUSB v.2.0.2, I added on usb_msc.cpp module this method:
void USBmscDevice::unmount(char *path, uint8_t lun) 
{
    vfs_fat_rawmsc_unmount(path, lun);
}

On diskio_rawmsc.cpp module I added:
void vfs_fat_rawmsc_unmount(char *base_path, uint8_t lun)  
{
    uint8_t pdrv = 0xff;
    for (size_t i = 0; i < FF_VOLUMES; i++)
    {
        if(ff_raw_handles[i] == lun)
        {
            pdrv = i;
            ff_raw_handles[i] = 0xff;
            break;
        }
    }
    
    if(pdrv < FF_VOLUMES) 
    {
        char drv[3] = {(char)('0' + pdrv), ':', 0};
        esp_vfs_fat_unregister_path(base_path);
        f_mount(0, drv, 0);
        ff_diskio_unregister(pdrv);
        ESP_LOGI(TAG, "unregister path: %s, volume pdrv: %s", base_path, drv);
    }
}

Then, on the example msc.ino inside
void client_event_callback(const usb_host_client_event_msg_t *event_msg, void *arg)
where actually it's managed USB_HOST_CLIENT_EVENT_NEW_DEV event
I added USB_HOST_CLIENT_EVENT_DEV_GONE event to unmount usb when usbdrive is unplugged by me:

...
else if (event_msg->event == USB_HOST_CLIENT_EVENT_DEV_GONE)
{
    if(device)
    {
      device->unmount(MOUNT_POINT, 0);
      delete(device);
      is_mount = false;
      ESP_LOGW(TAG, "Device unmounted");
    }
    device = NULL;
}

Event is triggered, but when I try to plug usb drive again it doesn't re mount it again.
Moreover, compared to the example you linked to me, I lack the deinit function.  
Am I working with the wrong library?
Regards

zenius06, 2023-06-23:
Hello, I hjave the same issue with the destructor, I need to modify usb_host.cpp to add the call back USB_HOST_CLIENT_EVENT_DEV_GONE like this :
void _client_event_callback(const usb_host_client_event_msg_t *event_msg, void *arg)
{
    USBhost *host = (USBhost *)arg;
    if (event_msg->event == USB_HOST_CLIENT_EVENT_NEW_DEV)
    {
        ESP_LOGI("", "client event: %d, address: %d", event_msg->event, event_msg->new_dev.address);

        if (host->_client_event_cb)
        {
            host->_client_event_cb(event_msg, arg);
        }
        else
        {
            host->open(event_msg);
        }
    }
    if (event_msg->event ==USB_HOST_CLIENT_EVENT_DEV_GONE)
    {
    	host->_client_event_cb(event_msg, arg);
    }
}

And 
void client_event_callback(const usb_host_client_event_msg_t *event_msg, void *arg)
  if (event_msg->event == USB_HOST_CLIENT_EVENT_NEW_DEV)
    {
...
}
else
{
    	ESP_LOGE("", "DEVICE gone event");

    	if(device)
    	{
			ESP.deepSleep(1);													// Workaround for restart after 1 secound, Workaround also for the USB destructor missing
			delay(100);															//

        }

It will be very useful if the futur version of this library V2_03 include a correct destructor like on IDF, again Chegewara thanks for for library