chegewara/EspTinyUSB #144: ESP32-S2 HID sendRelease not working

khorght, 2023-06-30:
MCU  - ESP32-S2
IDE - Arduino IDE 2.1.0
Library - EspTinyUSB 2.0.2

dev.sendRelease()  not working, keycode keeps appearing on Host PC Notepad forever after dev.sendKey(HID_KEY_A);

chegewara, 2023-07-03:
You have to add small delay between send and release, like 10ms or so.

khorght, 2023-07-04:
Hi, chegewara. I've tryed delay from 1 ms to 50 ms with no luck. Issue is not in sendRelease itself, only first report send is working and returning true and appears in wireshark, any other attempt to send report regardless of delay between them returns false, wireshark shows nothing. I'll try to investigate further, maybe, currently i've jumped to native esp-idf environment project, works as expected with esp-idf 5.1. Anyway thanx for your hard work.