chegewara/EspTinyUSB #145: Multiple key combinations?

deadshxll, 2023-07-04:
How can I press a key like Win+Shift+S? I know that I have to hold the GUI key first, then Shift, then R.

However when I send a key to hold down and then follow it up with another key it stops holding the first key making it not possible to do this

please write a POC example, would be very beneficial.

deadshxll, 2023-07-04:
I solved this by looking at the newer project and comparing it with this project, all I did was simply add a new function to do the stuff that I want.

I'll keep this issue up in case the developer has an answer.

chegewara, 2023-07-04:
You already found the answer.
In this library there is missing few features, it was simplified library. The other i started having some experience, but i have no time to continue.

deadshxll, 2023-07-05:
> You already found the answer. In this library there is missing few features, it was simplified library. The other i started having some experience, but i have no time to continue.

Awesome, thank you for creating this library though, saved my life.

closed 2023-07-05 by deadshxll