chegewara/EspTinyUSB #146: USB descriptor larger then expected in Arduino IDE

Lights-On-42, 2023-08-08:
Hello @chegewara,

I've been using your library, EspTinyUSB, to interface a USB camera with an ESP32-S2 module. First and foremost, I appreciate your work on developing this library, as it has been immensely helpful for USB mass storage functionality.

However, I've encountered an issue when attempting to connect my USB camera to the ESP32-S2. I've made sure to properly wire the connections, with the following pin configuration:

5V: 5 Volt
GND: Ground
D+: Connected to pin 19
D-: Connected to pin 20

While the library works well for USB mass storage, I'm facing an error specifically when trying to connect the USB camera. The error message I'm encountering is as follows:

E (26569) HUB: Configuration descriptor larger than control transfer max length
E (26569) HUB: Stage failed: CHECK_SHORT_CONFIG_DESC

After some investigation, I believe this issue might be related to the maximum length of the control transfer for the configuration descriptor. I'm wondering if it would be possible to address this by adjusting a value within the Arduino IDE.

I've attempted to modify the value of CHECK_SHORT_CONFIG_DESC by changing it from 1024 to 4096 using the following define:

#define CHECK_SHORT_CONFIG_DESC 4069
However, this modification did not yield the desired results and the issue persists.

Please look at this issue from Expressif
https://github.com/espressif/esp-idf/issues/9868

I wanted to reach out to you to seek your insights on this matter. Could you kindly provide guidance on how to resolve this issue? If there are any alternative solutions or adjustments that I should consider, I would greatly appreciate your input.

Thank you once again for your dedication to this library.

Best regards,
Lights on 42

serifpersia, 2023-10-14:
Were you able to fix this issue in Arduino IDE? I have similar issue on S3 USB MIDI Host, some devices work while some don't with this error. Any info you have is appreciated.

Lights-On-42, 2023-10-15:
Hello serifpersia,

I have looked in many files and found this code:

case ENUM_STAGE_CHECK_SHORT_CONFIG_DESC: {
        usb_config_desc_t *cfg_desc = (usb_config_desc_t *)(enum_done->transfer.data_buffer + sizeof(usb_setup_packet_t));
        full_config_length = cfg_desc->wTotalLength;
        UVC_CHECK_GOTO(full_config_length <= CTRL_TRANSFER_DATA_MAX_BYTES, "Configuration descriptor larger than control transfer max length", stage_failed_);
        break;

then i write in my Code 
#define CTRL_TRANSFER_DATA_MAX_BYTES 4096

The file name is usb_stream.h but it works not. I think the tinyUSB Version work with a other File.
And i find not the correct File witch used TinyUSB.

serifpersia, 2023-10-15:
Yeah I recompiled the sdk for s3 and s3, once you compile with new size for the max transfer you replace few files into the sdk mainly the root sdkconfig and qio_qspi/include/sdkconfig. I still have to test but i verified this changed the max transfer ie 1024 or something else from default 256

Lights-On-42, 2023-10-15:
Thanks,
i have change the sdkconfig file for my S2. And this problem works now.
And now i get this:
E (9746) HUB: Bad transfer status: CHECK_SHORT_DEV_DESC
E (9746) HUB: Stage failed: CHECK_SHORT_DEV_DESC

It is possply to make a new sdkconfig File in the arduino Folder?

serifpersia, 2023-10-15:
That does nothing. You need to recompile the whole sdk and replace it in arduino folders. I can build it for you and share a link if you want. Just confirm me are you in Arduino IDE, Windows or Linux, and ESP32 S2 or S3?

Lights-On-42, 2023-10-17:
Hello serifpersia,

i have a functionally esp-idf in visual studio code on windows and in this projekt it works on my esp32s3.
Can you please give me a explanation how this works?

https://github.com/espressif/esp32-arduino-lib-builder/tree/release/v4.4

can you help me for a ESP32S3 and my problem a usb web cam work on arduino?

serifpersia, 2023-10-18:
Since you already use esp-idf and have built working project in it, all you have to do is do this before building it. Launch ESP-IDF shortcut on desktop its like cmd thing, then cd to your project folder, you can back yup the folder just in case, then do idf.py set-target esp32s3, then idf.py menuconfig, go to component configuration, find usb otg, the first one should be max transfier size press enter and type 4096 default is 256, then ctrl s or s enter and it wil lsave the config in sdkconfig file, then you esc from this and you are back in that cmd windows, then you just do idf.py build, once done you can do flash via this command idf.py flash your-com-port-example-COM19 If you need help find me on discord serifpersia