chegewara/EspTinyUSB #148: Sending Hid Messages gives strange result

K372470, 2023-09-24:
# Good day

Im trying to make a controller for my device(NUX-MG30) and I need to send {12,192,4,88} to the usb interface of this device:
```
#include "hidgeneric.h"
#if CFG_TUD_HID
HIDgeneric dev;
const uint8_t buffer[4]={12,192,4,88};

void setup()
{
    Serial.begin(115200);
    dev.begin();
}

void loop()
{
    delay(1000);
    dev.write(buffer, sizeof(buffer));
}

#endif
```
This is a bit changed version of `hid_generic.ino`

# Problem

It works and sends data, BUT `USB Analyzer by Eltima Software` says im sending {01,12,192,4,88} => arduino adds '1' byte at the start of my message
Is there anything I can do with it?

# My settings

USB-MODE: USB-OTG (TinyUsb)
CDC-ON-BOOT: True
JTAG: Disabled

# End

Thanks to everyone who took a look at my problem

chegewara, 2023-09-26:
Its good that the 1 is added. It is report ID.
Its USB standard.

K372470, 2023-10-04:
Hmm...
Sounds realistic.
Could you then give me any tip for "is my idea is even realisable",please ?
As i can think esp32 should programmed as usb-host, but when i tried no data was sent at all(

chegewara, 2023-10-06:
There is a chance it may works when report ID is 0.
Please check code how to set report ID.