chegewara/EspTinyUSB #150: Adding a new USB device which does not fit into the included classes.

JIGR1969, 2023-10-16:
I am looking at creating a IoT device which will have it's own vendor and product ID, but who's description does not fit in with the included defined device classes. Basically I want to fool another device into thinking that it has a USB printer attached to it.

Where do I start? 

I could have a defined class in a CPP and H files in the same directory as the project (coding in Arduino IDE), but couldn't find a good example of how the vendor and product ID are defined. 

Any help would be gratefully received. Thanks

chegewara, 2023-10-17:
This is working for any device class
https://github.com/chegewara/EspTinyUSB/blob/master/examples/device/basic_setup/basic_setup.ino#L32-L36

JIGR1969, 2023-10-19:
Hi Chegewara.

That is the code I've started with and made some good progress with it. 

However in order to get the device descriptors below changed from what they would be (shown below):

Device Descriptor:
bcdUSB:             0x0200
bDeviceClass:         0xEF
bDeviceSubClass:      0x02
bDeviceProtocol:      0x01
bMaxPacketSize0:      0x40 (64)

To what I need them to be:

Device Descriptor:
bcdUSB:             0x0200
bDeviceClass:         0x00
bDeviceSubClass:      0x00
bDeviceProtocol:      0x00
bMaxPacketSize0:      0x40 (64)

I've amended, for the time being, the code below found in the "usb_descriptors.cpp" file. Which isn't ideal as this will be overridden at the next update of the library.

    if (enableCDC) {
        // Use Interface Association Descriptor (IAD) for CDC
        // As required by USB Specs IAD's subclass must be common class (2) and protocol must be IAD (1)
        _descriptor_config.bDeviceClass = 0x00;
        _descriptor_config.bDeviceSubClass = 0x00;
        _descriptor_config.bDeviceProtocol = 0x00;
    } 

I'd like to be able to set these values within the project as opposed to relying on the source files. I'm still looking at where the configuration descriptors below are set, and obviously would like to configure these within the project as opposed to the source files:

Configuration Descriptor:
wTotalLength:       0x004B
bNumInterfaces:       0x02
bConfigurationValue:  0x01
iConfiguration:       0x00
bmAttributes:         0xA0 (Bus Powered Remote Wakeup)
MaxPower:             0xFA (500 Ma)

Any help/advice/pointers you can put my way would be gratefully received.

Regards
James.

chegewara, 2023-10-20:
It is arduino library, which means it was meant to use by newb-ish users.
The library is far from being good and its missing option to set few values.

My suggestion is to fork library and to add options you need, or just change values you mentioned.

ggg-g, 2023-12-27:
我也在尝试使用esp32s3，让它被计算机识别成打印机，通过USB进行连接，期待作者支持