chegewara/EspTinyUSB #151: MIDI problem

test1025, 2024-01-06:
Under Windows 11, MIDI can be recognized by the device manager, but MIDI messages are not seen in various DAW software and MIDI-OX

test1025, 2024-01-06:
I have also tried using a mobile phone, which can recognize MIDI devices, but there is no MIDI message output

erb997, 2025-01-12:
same thing on Win10
I can connect and show midi.product name of my choosing. midi.manufacturer doesn't show in device detail. GrandOrgue can see the device but doesn't show any events in log or play any notes. Did this ever get solved? I am using an ESP32-S3 board.

infrafast, 2025-05-19:
hello my device esp32-s3 (seed xiao) is also seen correectly as midi device under  windows  11 and macos but no notes are catched by any software. Any help appreciated on how you solved it.

test1025, 2025-05-19:
> hello my device esp32-s3 (seed xiao) is also seen correectly as midi device under windows 11 and macos but no notes are catched by any software. Any help appreciated on how you solved it.

Actually, I haven't been successful yet.My problem phenomenon is completely the same as yours...TxT...

infrafast, 2025-05-19:
> hello my device esp32-s3 (seed xiao) is also seen correectly as midi device under windows 11 and macos but no notes are catched by any software. Any help appreciated on how you solved it.

Actually I have added midi.setBaseEP(2); in the setup() and it now works for me

hanbingniao1, 2025-05-19:
> > hello my device esp32-s3 (seed xiao) is also seen correectly as midi device under windows 11 and macos but no notes are catched by any software. Any help appreciated on how you solved it.
> 
> Actually I have added midi.setBaseEP(2); in the setup() and it now works for me

Wow, thank you for sharing! I will try it out later