chegewara/EspTinyUSB #152: Remote pendrive code not working on ESp32 s3

micromagine, 2024-02-16:
examples/host/remote_pendrive
i tried this code on esp32 s3 compiled ok but when insert pendrive to esp32s3 device get reset

my errors are
ELF file SHA256: 0cf01760fb528cb7

micromagine, 2024-02-16:
[image]

micromagine, 2024-02-16:
when i insert pendrive it reset continues other wise it works !! my task is i want to take data log in pendrive !! so m i doing anything wrong or pendrive is formatted in FAT. is it wrong !!! do you have any detail documentation or video for this code !!!

micromagine, 2024-02-16:
I am eagerly anticipating your response! I am somewhat uncertain about the capability of connecting a pendrive to the ESP32-S3 and performing read and write operations for data logs. Your code suggests that this is achievable, and I would greatly appreciate your assistance in clarifying this matter.

micromagine, 2024-02-19:
hello sir

chegewara, 2024-02-20:
https://github.com/espressif/esp-usb/tree/master/host/class

micromagine, 2024-02-20:
thank you it works!!!!

micromagine, 2024-02-21:
https://github.com/nathalislight/esp32s3_USB_Drive/tree/main

similar code for Arduino Esp32-s3 USB HOST MSC is available .
you can update your code !!! thanx for help 
i am posting this because i am searching from this last one year now i get solution !!! thanx to https://github.com/nathalislight/

closed 2024-02-22 by chegewara