chegewara/EspTinyUSB #153: Release only one of the keys

alonso240, 2024-03-30:
Hello.

First of all, thank you very much for the fantastic libraries.

I'm trying to make an ESP32-S3 behave as a keyboard+mouse.

Everything works correctly, but I have a question.

I want that when I hold down a button, the keyboard key is also pressed, I achieve that with "sendPress".

If, for example, I hold down the "A" and "S" keys, I release the "S" key and hold down the "A" key. How do I do it?

If I execute "sendRelease", as I understand it stops pressing all the keys.

Sorry for my English from Google Translate!!!

Thank you so much!!!!

Greetings.

chegewara, 2024-03-31:
Hi,
when 2nd time you send `sendPress()` with only `A` OS should interpret it like you released `S`

alonso240, 2024-04-01:
So:

If I send "sendPress(A)" and then send "sendPress(B)" it would not add the presses (A and B).

I deactivate "A" and press "B"?

chegewara, 2024-04-02:
Yes