chegewara/EspTinyUSB #154: HIDgeneric host-side example?

IU5HKU, 2024-04-03:
The library works great for me, all the examples compiles and works as expected with ESP32-S2, but i don't understands how to set S2 when some HIDgeneric client is connected to usb port. I've read a lot and found that the raw communications is done through reports, so you need a reportID...when you are a client...you initialize the device and the library under the hood makes the work for you, so you end in a simple write() to communicate with the host, the callback routine will provide data who comes back from the host, and this is ok. But if S2 act as host? How can i "open" a client to communicate in raw mode by means of reports? Please could you point me to the solution of this particular case? Thank you so much for your time.
Regards
 Marco

chegewara, 2024-04-03:
It is actually very simple.
First byte of raw data is reportID.
Thats all.

IU5HKU, 2024-04-03:
So when i check the bInterfaceClass for the HIDgeneric, i don't need to initialize the just connected client device in some way? As per ACM or MSC examples? I don't need endpoints in this case, bit i've not understood how i can retrieve or send data without tell to the device object where to put it, there's no need to a callback routine for data handling host-side? Sorry, i know, i'm quite a newbie in the usb field...

chegewara, 2024-04-04:
Making this library was my learning curve, and i am still newbie.

Yes, of course you need to get descriptor etc like in ACM or MSC, but i dont remember now how to do it exactly (i dont work with USB for a year or so).

IU5HKU, 2024-04-04:
Never mind, thank you so much anyway, i'll keep searching around and reading...:-)
Have a nice day!

IU5HKU, 2024-04-06:
Never mind, thank you so much anyway, i'll keep searching around and reading...:-)
For who need to read something about how USB works i suggest this link:

https://www.keil.com/pack/doc/mw/USB/html/_u_s_b__concepts.html

is simple and well explained, it only scratch the surface, but isn't bad at all to get in touch with USB.
Have a nice day!