chegewara/EspTinyUSB #157: hal/usb_hal.h missing in Arduino

mortenp, 2024-06-23:
For both ESP32S3 andESP32S2, on both MacOS and Windows, I am missing usb_hal.h
Boards are up to date with esp32 boards v. 3.0.1

This is the only library where I get this compilation error, but I have no idea whether it's a dependency on another library, or a file I'm supposed to find somewhere else.

/Arduino/libraries/ESP32TinyUSB/src/esptinyusb.cpp:3:10: fatal error: hal/usb_hal.h: No such file or directory
    3 | #include "hal/usb_hal.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1

chegewara, 2024-06-24:
Hi,
probably my library is not compatible with arduino-esp32 v3.x. 
Im guessing that this header file change name in esp-idf 5.x.

sebcbien37, 2024-08-01:
Hi, 
Will you update your library for esp32 v3 ? It could help me.
Thanks.

chegewara, 2024-08-01:
Not in the near future, sorry.

imakin, 2024-09-09:
hi, i got the same error in that new library https://github.com/chegewara/esp32-usb-v2/issues/12

LJH666666666, 2024-11-09:
Hi, has anyone solved this issue yet? I really need a solution.

sebcbien37, 2024-11-12:
No, I always use the ESP32 lib in 2.0.17

MrMabulous, 2025-03-08:
Ran into the same issue. Anyone knows a solution?

Steffen-W, 2025-04-10:
The following settings work for me with pio:

```
board = esp32-s2-saola-1
framework = arduino
platform = espressif32@5.1.1
board_build.mcu = esp32s2
```

espressif32@5.2.0 and newer are not working.

Steffen-W, 2025-04-11:
The best thing for me now is to switch to TinyUSB. I couldn't use it in my old code because it wasn't available yet. But now it is available.

```
[env:esp32-s2-saola-1]
board = esp32-s2-saola-1
framework = arduino
board_build.arduino.usb = native
platform = espressif32@6.10.0
board_build.mcu = esp32s2
```

```cpp
#include "USB.h"
#include "USBCDC.h"

USBCDC USBSerial;

void setup() {
  USB.begin();
  USBSerial.begin();

  while (!USBSerial) delay(100);

  USBSerial.println("Native USB Serial");
}

void loop() {
  USBSerial.println("Ping!");
  delay(1000);
}
```